tscircuit/runframe #271: Bad font sizes and confusing messages on RunFrameForCli

seveibar, 2025-02-19:
[image]

- Use `text-xs` everywhere (high density)
- DO NOT say "Unsaved Changes", say "Not Synced", "Not Synced" should be clickable
- Show the error when something fails to save in the browser, don't tell the user to go to the console (the CLI can emit the error as an event on the event bus)

/bounty $15

CC @ArnavK-09

MustafaMulla29, 2025-02-19:
/attempt #271

MustafaMulla29, 2025-02-19:
[image]

@seveibar There is no message in the error object, what should I show the error?

seveibar, 2025-02-19:
Also RE:text, we should give the full names for each export type, e.g. gbr -> "Gerbers", dsn -> "Specctra DSN", etc.

seveibar, 2025-02-19:
@MustafaMulla29 discussed on discord a bit with arn but basically the CLI needs to be updated to give messages and otherwise what arn originally wrote is fine, but it should be in a dialog.

also fwiw we should let Arn get dibs on this issue if he can- there are plenty of other things related to this (you can get github sponsored for creating issues that get bountied btw- a lot of contributors get a lot of $$ just via finding great issues and sometimes fixing the issues they find!)

MustafaMulla29, 2025-02-19:
@seveibar oh ok, so should I cancel my attempt?

ArnavK-09, 2025-02-19:
/attempt https://github.com/tscircuit/runframe/issues/271

closed 2025-02-21 by imrishabh18