tscircuit/runframe #463: On moving the components around it should not snap back to the original position

imrishabh18, 2025-04-15:
Similar to `<PcbViewer />` on moving the components, the component flows around which the drag. Unlike in here we get a snap back to the original position, and then after the render we do get the new position.

Refer [PcbViewer](https://github.com/tscircuit/pcb-viewer) for this.

https://github.com/user-attachments/assets/26578eda-275b-4c82-8ca8-88b3c79c4f59

/bounty $10

Anshgrover23, 2025-04-15:
@imrishabh18 https://github.com/tscircuit/runframe/issues/387 same issue exists?

Abse2001, 2025-04-17:
/attempt #463

Abse2001, 2025-04-18:
@imrishabh18 I think this issue is solved?

https://github.com/user-attachments/assets/0fa5961f-3f27-44b4-b8ac-7044b69e9f5e

imrishabh18, 2025-04-18:
Yeah @Abse2001 the fix got merged yesterday, thanks for pointing it out. Closing it

closed 2025-04-18 by imrishabh18