tscircuit/runframe #1135: "Open File" dropdown improvements, show current folder content + show nested items as collapsible folder

seveibar, 2025-09-13:
You will need to create a new example, similar to example05.page.tsx but with more files to test

[image]

User request:
[image]

/bounty $100

AnasSarkiz, 2025-09-13:
@seveibar,  something like this?
[image]

seveibar, 2025-09-13:
I dont like the file icons for example, you may want to discuss variations of the design with screenshots on discord. I think your marginX is too wide as well

PR #1169 (2025-09-19): feat(fileSelector): improve open file dropdown
- When you type, it prioritizes files in your current directory at the top
- Files from other directories show below with their full paths
- Search clears automatically when you close the dropdown
- No more clutter from section titles since the breadcrumbs provide sufficient context

https://github.com/user-attachments/assets/289f0a2f-82a4-434b-af66-d351cc400bb9

fix #1135 
/claim #1135
closed 2025-09-19 by seveibar