tscircuit/runframe #1180: Introduce `<RunFrameStaticBuildViewer />`

seveibar, 2025-09-19:
This is the _exact same UI as `<RunFrameForCli />`_ but it works on a directory of prebuilt circuit json files.

This is intended to be used on tscircuit.com to allow viewing a project that has been prebuilt with `tsci build`. It replaces the version of RunFrame at https://tscircuit.com/seveibar/led-water-accelerometer/releases/8b90619f-14ef-4c5a-89f1-2476566c08a5/preview which currently rebuilds in browser incorrectly.

The user should be able to do File > Export to export still. When they hit export, it will do the regular/same logic to convert Circuit JSON- hopefully no modification needed there.

Code reuse is critical. Some code may need to be adapted.

Work is complete when integrated to the tscircuit.com page above

CC @ArnavK-09 @ShiboSoftwareDev @imrishabh18 

/bounty $150

Harsh9485, 2025-09-19:
/attempt #1180

vishwamartur, 2025-09-19:
/attempt #1180

PR #1182 (2025-09-19): feat: add RunFrameStaticBuildViewer component
fix #1180